adafruit/ci-arduino #4: Allow Arduino Version as a parameter

chatelao, 2015-07-30:
To provide some regression testing, a compilation with a new and an old version should be possible to recognize fails due to dependency changes.

toddtreece, 2015-07-30:
@chatelao are you referring to changing between 1.0.x to 1.6.x versions? or mainly between 1.6.x versions?

chatelao, 2015-07-31:
Whatever version will come in the future. I would like to set the version myself, something like:
- build_platform uno 1.6.4
- build_platform uno 1.6.5
- build_platform uno latest

In case a new version fails, I would like to prove that an old version worked fine and find the difference in the Arduino package.

toddtreece, 2015-07-31:
I think this is probably overkill for our purposes since we are mostly concerned with testing current stable version of the IDE. Arduino should be testing IDE regressions in their own repo, and it would add a lot of extra runtime to check multiple IDE versions in our tests.

I'm going to close this one since we aren't going to be adding it, but feel free to issue a pull request if you end up implementing it and we can look at merging it in.

Thanks!

closed 2015-07-31 by toddtreece

chatelao, 2015-08-01:
Thanks for the fast feedback.