adafruit/ci-arduino #8: fatal error: WiFiUDP.h: No such file or directory

probonopd, 2015-08-15:
I constantly get `fatal error: WiFiUDP.h: No such file or directory` when I try to compile my sketch using this. 

```
fatal error: WiFiUDP.h: No such file or directory
compilation terminated.
Error compiling.
```

[Here](https://github.com/probonopd/ESP8266HueEmulator/blob/master/.travis.yml) is my .travis.yml, and [here](https://s3.amazonaws.com/archive.travis-ci.org/jobs/75759866/log.txt) is the full log.

chatelao, 2015-08-16:
Did you install a special library to your local Arduino IDE to get it running on your machine?

probonopd, 2015-08-16:
No; as far as I know it is bundled with the esp8266 platform. https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiUdp.h

tdicola, 2015-08-16:
Are you running your sketch on a case insensitive file system like Mac OSX?  It looks like the header from the ESP8266 core library is called WifiUdp.h: https://github.com/esp8266/Arduino/tree/esp8266/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src  It likely works on your machine if the filesystem isn't case sensitive, but then fails when travis runs it on a little Linux VM (with a case sensitive FS).  Try changing the sketch / example code to use the WifiUdp.h name instead of WifiUDP.h.

closed 2015-08-16 by tdicola

probonopd, 2015-08-17:
Thank you. I had missed the obvious, exactly as you described. It is working now. Thanks for this useful project.