adafruit/ci-arduino #11: Travis CI today

thinkyhead, 2016-09-20:
Since earlier today our specific project, MarlinFirmware, has been experiencing a sudden "outage" with Travis CI as it tries to download and install u8glib. Have you also seen this, and do you know what might be causing it?
```diff
  Downloading library: U8glib Downloaded 1,153kb of 1,153kb.
  Verifying archive integrity...
- java.lang.Exception: CRC doesn't match. File is corrupted.
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:77)
	at cc.arduino.contributions.libraries.LibraryInstaller.install(LibraryInstaller.java:97)
	at processing.app.Base.<init>(Base.java:414)
	at processing.app.Base.guardedMain(Base.java:222)
	at processing.app.Base.main(Base.java:135)
-  The command "arduino --install-library "U8glib"" failed and exited with 255 during .
```
https://travis-ci.org/MarlinFirmware/Marlin/builds/161188040

toddtreece, 2016-09-20:
try re-running the build. i was getting this error locally as well while testing `U8glib`, so it's not travis ci specific. it seems fine now, so my guess is there was a temporary glitch in the arduino library index that caused checksums to fail.

closed 2016-09-20 by toddtreece

thinkyhead, 2016-09-21:
Seems to be the case. I notified the Arduino crew, but they were probably on top of it before then. In our own project I just switched to downloading it straight off Github instead of using `--install-library` and that did the trick.