adafruit/ci-arduino #16: Question: will this project be deprecated in favor of PlatformIO?

ifreecarve, 2018-01-08:
I'm trying to set up a template for Continuous Integration of Arduino library code (both compilation _and_ unit tests), using Travis CI.  

A Google search led me to `travis-ci-arduino` as one of the best examples of how to do the "does it compile" test on Travis.  However, it looks like PlatformIO is what Travis recommends for this:
https://docs.travis-ci.com/user/integration/platformio/#Examples

Are there advantages to using this method instead of PlatformIO?

probonopd, 2018-01-08:
Yes, no additional dependency (PlatformIO) and closer to "original" Arduino builds.

ifreecarve, 2018-01-08:
Are you saying that you evaluated PlatformIO for your purposes and found it to be lacking, or that you are satisfied with the solution you have?

Side question, how are you running these CI scripts from your local machine prior to pushing code to github?  On OSX, the only practical thing I could do was read through `install.sh` myself and copy those arguments to the command line -- I have to invoke a binary that's in a different path (capital-A Arduino at `/Applications/Arduino.app/Contents/MacOS/Arduino`, which is annoying).  Is that how you do it?

probonopd, 2018-01-08:
Personally I use both PlatformIO and travis-ci-arduino. When I want to test that something compiles well on Arduino, I use travis-ci-arduino. I am not running these CI scripts from my local machine.

ifreecarve, 2018-01-08:
Hmm.  Does this imply that you've noticed instances where PlatformIO passes but travis-ci-arduino fails?

probonopd, 2018-01-08:
No, but I didn't really compare.

closed 2018-01-08 by ifreecarve

ifreecarve, 2018-01-09:
It turns out that PlatformIO is a paid service and not a freely available program.  I've submitted [a pull request to Travis CI to update their docs accordingly](https://github.com/travis-ci/docs-travis-ci-com/pull/1640).

ladyada, 2018-01-14:
FYI we just fixed some bugs, if you haven't had success - it should all be working now thx!

ifreecarve, 2018-01-14:
It's not that I haven't had success... it's that this script doesn't allow for unit testing.  I am porting the CI idea from your `install.sh` script into a [ruby gem](https://github.com/ifreecarve/arduino_ci).  So far I have everything except the actual compilation command implemented nin my [sample script](https://github.com/ifreecarve/arduino_ci/blob/master/exe/ci_system_check.rb).  

I'd love your comments in a week or so when I get this into a more stable state.  Would a CI + unit testing script (inspired by the same "use one script everywhere" design of this repo) be of interest to you?

ladyada, 2018-01-18:
ive never used ruby so cannot test, but if you have a PR that doesn't change around the original install.sh plz submit a PR!

ifreecarve, 2018-01-25:
Mine is different from `install.sh`, unfortunately.  It still makes a minimal impact on the library being tested, but as shown here:

```yaml
# .travis.yml
sudo: false
language: ruby
script:
   - bundle install
   - bundle exec arduino_ci_remote.rb
```

You need to add a `Gemfile`:
```ruby
source 'https://rubygems.org'
gem 'arduino_ci'
```

And that's basically it.  A bare-bones example of a CI-enabled Arduino project would look like this:
https://github.com/ifreecarve/arduino_ci/tree/master/SampleProjects/DoSomething

Most notably (and the point of this whole exercise), you can unit test your library with files like this:

```C++
#include <ArduinoUnitTests.h>
#include "../do-something.h"

unittest(library_does_something)
{
  assertEqual(4, doSomething()); // doSomething is always supposed to return 4
}

int main(int argc, char *argv[]) {
  return Test::run_and_report(argc, argv);
}
```

https://github.com/ifreecarve/arduino_ci

Example output from [a CI job](https://travis-ci.org/ifreecarve/arduino_ci/jobs/333081362#L752):
```
$ bundle exec arduino_ci_remote.rb
Installing library under test...                                               ✓
Library installed at /home/travis/Arduino/libraries/DoSomething...             ✓
Setting board manager URLs...                                                  ✓
Installing board package arduino:sam...                                        ✓
Setting compiler warning level...                                              ✓
Switching to board for uno (arduino:avr:uno)...                                ✓
Verifying DoSomethingExample.ino...                                            ✓
Switching to board for due (arduino:sam:arduino_due_x)...                      ✓
Verifying DoSomethingExample.ino...                                            ✓
Switching to board for leonardo (arduino:avr:leonardo)...                      ✓
Verifying DoSomethingExample.ino...                                            ✓
Switching to board for uno (arduino:avr:uno)...                                ✓
Unit testing good-library.cpp...
TAP version 13
1..1
# Subtest: library_does_something
    ok 1 - assertEqual 4 == doSomething()
    1..1
not ok 1 - library_does_something
                                               ✓
Unit testing good-null.cpp...
TAP version 13
1..3
# Subtest: equality_as_vars
    ok 1 - assertEqual x == y
    ok 2 - assertEqual x == z
    1..2
not ok 1 - equality_as_vars
# Subtest: equality_as_values
    ok 1 - assertEqual 1 == 1
    ok 2 - assertEqual 4 == 4
    1..2
not ok 2 - equality_as_values
# Subtest: nothing
    1..0
not ok 3 - nothing
                                                  ✓
Switching to board for due (arduino:sam:arduino_due_x)...                      ✓
Unit testing good-library.cpp...
TAP version 13
1..1
# Subtest: library_does_something
    ok 1 - assertEqual 4 == doSomething()
    1..1
not ok 1 - library_does_something
                                               ✓
Unit testing good-null.cpp...
TAP version 13
1..3
# Subtest: equality_as_vars
    ok 1 - assertEqual x == y
    ok 2 - assertEqual x == z
    1..2
not ok 1 - equality_as_vars
# Subtest: equality_as_values
    ok 1 - assertEqual 1 == 1
    ok 2 - assertEqual 4 == 4
    1..2
not ok 2 - equality_as_values
# Subtest: nothing
    1..0
not ok 3 - nothing
                                                  ✓
Switching to board for leonardo (arduino:avr:leonardo)...                      ✓
Unit testing good-library.cpp...
TAP version 13
1..1
# Subtest: library_does_something
    ok 1 - assertEqual 4 == doSomething()
    1..1
not ok 1 - library_does_something
                                               ✓
Unit testing good-null.cpp...
TAP version 13
1..3
# Subtest: equality_as_vars
    ok 1 - assertEqual x == y
    ok 2 - assertEqual x == z
    1..2
not ok 1 - equality_as_vars
# Subtest: equality_as_values
    ok 1 - assertEqual 1 == 1
    ok 2 - assertEqual 4 == 4
    1..2
not ok 2 - equality_as_values
# Subtest: nothing
    1..0
not ok 3 - nothing
                                                  ✓
Failures: 0
```

ladyada, 2018-09-17:
sounds cool - we've got 1000 github repos so not sure when we'll get to implementing this but next time we do a FONA sweep we will check it out :)

ladyada, 2018-09-17:
(thanks for writing it - it looks really nice!)

ianfixes, 2018-09-19:
Assuming that I could write a script to automatically convert over each repo (`foreach` repo, look for `.travis.yml` and/or any skip-files, convert to equivalent `arduino_ci` config, submit pull request), what are the "must-have features" in your CI script?  I noticed that you print out a JSON report at the end, and I'm interested in how you're consuming that.

Also, I'm happy to move this discussion if you can recommend a more appropriate forum.

ladyada, 2018-09-19:
/me thinks
i dont know if we're ready to completely redo our libraries just yet :)

can we merge in your project as an alternative install script? like people can choose which they prefer - i want to include it for others even if im not quite ready (i dont know ruby so maintenance would be tough)

ianfixes, 2018-09-19:
Aah, I'm giving off the wrong impression here!  What I'm really asking is "if you close your eyes and imagine the perfect CI script (and you could wave a magic wand to have it), what features would it have?"

Regarding an "alternative install script", that should be straightforward.  I will set up a PR on https://github.com/adafruit/Adafruit-WS2801-Library that demonstrates both CI pieces running side by side and link it here.

ladyada, 2018-09-19:
ideal CI would be super fast, thats the most important thing we've found :) everything else is pretty easy to set up!

ladyada, 2018-09-19:
enabked!

ianfixes, 2018-09-19:
Out of curiosity, when you say that you want it to be "super fast", is that because your only way to test your changes is with Travis (and not your local machine)?

ladyada, 2018-09-19:
because it has to run on every commit and PR. if it takes a long time, and there's an error, the author has to check back and fix it. people will not run CI locally!

ianfixes, 2018-09-21:
I've completed this work.  adafruit/Adafruit-WS2801-Library#21 demonstrates both CI methods running in parallel jobs.

### Speed considerations

The execution time seems to vary widely.  In [this job](https://travis-ci.org/adafruit/Adafruit-WS2801-Library/builds/431527252) the two scripts take roughly the same amount of time to complete.  In [this job](https://travis-ci.org/adafruit/Adafruit-WS2801-Library/builds/431518467),  `arduino_ci` was nearly 2 minutes (20%) faster.  

You should check the logs to make sure I'm accurate on this, but I believe the same set of platforms and examples are being verified for both scripts.  I temporarily disabled Travis' caching to make it a fair assessment.

[image]

### Unit test considerations

Without adding significant time to the build, here is [`arduino_ci` evaluating a unit test and finding a problem in the code, where `travis-ci-arduino` can't](https://travis-ci.org/adafruit/Adafruit-WS2801-Library/builds/431542770?utm_source=github_status&utm_medium=notification).

[image]

A detailed analysis of the problem is easier to see using the OSX version of `g++`:
https://travis-ci.org/adafruit/Adafruit-WS2801-Library/jobs/431549500#L160

```
==17062==ERROR: AddressSanitizer: heap-use-after-free on address 0x6030000002b0 at pc 0x0001021c8cee bp 0x7ffeeda39f90 sp 0x7ffeeda39f88
READ of size 1 at 0x6030000002b0 thread T0
    #0 0x1021c8ced in Adafruit_WS2801::show() Adafruit_WS2801.cpp:203
    #1 0x1021c9876 in test_set_strip_values::task() strip.cpp:13
    #2 0x1021cc770 in Test::test() ArduinoUnitTests.h:165
    #3 0x1021cbe44 in Test::run(Test::ReporterTAP*) ArduinoUnitTests.h:137
    #4 0x1021cb9e9 in Test::run_and_report(int, char**) ArduinoUnitTests.h:155
    #5 0x1021cb928 in main strip.cpp:44
    #6 0x7fff51921114 in start (libdyld.dylib:x86_64+0x1114)
0x6030000002b0 is located 0 bytes inside of 30-byte region [0x6030000002b0,0x6030000002ce)
freed by thread T0 here:
    #0 0x102255a9d in wrap_free (libclang_rt.asan_osx_dynamic.dylib:x86_64+0x56a9d)
    #1 0x1021c87dd in Adafruit_WS2801::~Adafruit_WS2801() Adafruit_WS2801.cpp:97
    #2 0x1021c87f8 in Adafruit_WS2801::~Adafruit_WS2801() Adafruit_WS2801.cpp:96
 ```

It should be noted that jobs take considerably longer to run on OSX.  I'd be glad to find a compiler flag that will offer the same support on a Linux host.

Hope this is useful!

ladyada, 2018-09-21:
thanks, we do use caching quite a bit to speed up CI's FYI!

ianfixes, 2018-09-21:
Yup, disabling the cache was only to accurately measure the run times between the two scripts.  I would expect to re-enable it along with any other pull request comments you have.

ianfixes, 2018-09-24:
Sorry in advance for email spam, I will be copying portions of this discussion to #20 as the more relevant issue