adafruit/ci-arduino #20: For your consideration: Enabling unit tests on Travis CI using a different script

ifreecarve, 2018-03-07:
Context: https://github.com/adafruit/travis-ci-arduino/issues/16

I've created an [alternative CI system](https://github.com/ifreecarve/arduino_ci) to the `install.sh` script found in this repository.  This was a personal project and I make no claim that it is better for your purposes than what you have now.  However, since the development was heavily inspired by `install.sh` and the proof of concept was built against [an Adafruit library](https://github.com/adafruit/Adafruit_FONA/pull/91), it seems appropriate to submit my work for your consideration.

## Who Cares About Unit Tests?

Unit testing capabilities give Travis CI the ability to validate the logic as part of the pull request process.

Although the existing `install.sh` script is great for verifying that your library will compile across a selected number of platforms, it says nothing about whether the logic is correct.  This means that accepting a PR from a stranger on GitHub means either mentally debugging their code by looking at the diff, or the more reliable / more labor intensive process of uploading that code on a board and testing it yourself by trying to put it in every possible hardware/software state.

By writing unit tests (which include the ability to apply fake digital/analog inputs, verify digital/analog outputs, read and write to serial ports, simulate behavior of external devices, etc), you can prove the correctness of your code without any wiring (and for that matter, without having a physical board at all).  You can simulate common and corner cases with the same relative ease.

Of course, you can't just compile standalone test programs with the Arduino IDE, let alone control the input states of physical pins.

## What Do You Propose?

It's simple, we fake the entire Arduino library.

## If It's So Simple, Why Hasn't Someone Done It Already?

If you're good at something, never do it for free.

J/K, this is a freedom-free alternative to PlatformIO -- Apache-2.0-licensed source and no signup nonsense required.  And it wasn't simple, it was actually kind of time consuming to get a build environment that would do both Arduino-IDE-based and GCC-based compilations, on both my OSX machine and on Travis CI's linux images.  But the dirty work is done and compilation seems to be reliable (I tested with Apple's `clang`-based `g++`, gcc-7, and gcc-4.9 (plus the gcc-4.8 available on Travis).  Most (but not all) builtin Arduino libraries are mocked for testing purposes, and adding more is straightforward.  

## Why Trust A 2-month-old Software Project With All Our CI?

You shouldn't.  I can't claim to know enough about your needs to say that this will serve them.  What I can say is that [the `arduino_ci` ruby gem is pretty well tested](https://travis-ci.org/adafruit/Adafruit_FONA/builds/350489990) and more test cases are easy to add.  In fact, I would appreciate being sent some test cases that break my CI module.

## Why Ruby?

It's a cross-platform scripting language that isn't a nightmare when it comes to package dependencies.

ianfixes, 2018-05-06:
Brief update: as of v0.1.10, [arduino_ci](https://github.com/ianfixes/arduino_ci) is now fully automated across Windows, OSX, and Linux hosts.  You can write your tests under one OS and have them run and pass under another OS with the same outcomes.

Crucially, you can run the unit tests both locally and remotely with the same script.

scls19fr, 2019-01-01:
Any news? This is part of a more general issue : having better quality for Arduino libraries
https://github.com/arduino/Arduino/issues/7567

ladyada, 2022-02-11:
closing since we're not supporting travis anymore

closed 2022-02-11 by ladyada