adafruit/ci-arduino #44: Strange behaviour when installing multiple libraries

leonardoInf, 2019-07-05:
When you have multiple dependencies in your project, you have to install them in seperate commands, otherwise it will not work

works: 
   - arduino --install-library "Adafruit Unified Sensor"
   - arduino --install-library "Adafruit GFX Library"

doesn't work:
  - arduino --install-library "Adafruit Unified Sensor","Adafruit GFX Library"

It seems to be an issue with the script and not arduino itself, because usually a list of libraries is accepted.

ladyada, 2019-07-07:
yeah, i dont know why it doesnt work. if you can figure it out please submit a PR :)

informatik-mueller, 2019-07-12:
I see that TravisCI is probably the issue here. I don't think there's much you can change in this repo. I will just submit a little change in the README so new users will not be disappointed.

leonardoInf, 2019-07-12:
I propose [this](https://github.com/adafruit/travis-ci-arduino/pull/45) (pull request)

ladyada, 2019-07-12:
please make a PR just for this one fix, and you can propose others in separate PRs!

leonardoInf, 2019-07-12:
agreed

ladyada, 2019-07-12:
done :)

closed 2019-07-12 by ladyada