adafruit/ci-arduino #55: Include/exclude tested sources

suculent, 2019-11-29:
I have private repository full of various .ino files, but only one folder (./src) is valid for testing. Or let's say I'd like to add ./test to have all the tests implemented there.
How do I exclude other files? Or how do I set build root?

Sorry for using GitHub for a FAQ.

ladyada, 2019-11-29:
hiya you'd probably need to adapt https://github.com/adafruit/travis-ci-arduino/blob/master/install.sh and make your own version!

closed 2019-11-29 by ladyada

suculent, 2019-11-30:
Seems I've found a more straightforward solution to original question (a bit outside the box).
All it needs is to `- cd ./src` or `- cd ./test` in the CI script before running this `build_main_platform` script in order to divide code forks as needed...

ladyada, 2019-11-30:
great!