adafruit/ci-arduino #59: build_main_platforms and build_cplay_platforms commands not found

makermelissa, 2019-12-17:
While pushing a PR to Travis. It fails with the following errors:
$ build_cplay_platforms
build_cplay_platforms: command not found
The command "build_cplay_platforms" exited with 127.
$ build_main_platforms
build_main_platforms: command not found
The command "build_main_platforms" exited with 127.

I noticed this on another PR that had failed for the same reason. Since this repo is the source of those commands, it seemed appropriate to file this here.

ladyada, 2019-12-17:
hmm ok please revert the last PR, seemed like it should have worked but did not

makermelissa, 2019-12-17:
Just confirmed that reverting it fixed this.

closed 2019-12-17 by makermelissa

BenoitAnastay, 2019-12-17:
I can investigate the issue, how where you running the script 
Surely you forgot to update the link to the build,

thoses comments are no more sourced, I can fix it but it works if you call `$TRAVIS_BUILD_DIR/install.sh build_platform esp8266` for example

makermelissa, 2019-12-17:
This is being called in automated builds in several hundred libraries. Can this be updated in a way so it doesn't break all these libraries?

BenoitAnastay, 2019-12-17:
Done,
I was trying to not doing something messy but I didn't thought that may brake previous installs, so now the functions are sourced for travis AND callable through argument for Github