adafruit/ci-arduino #74: doxy_gen_and_deploy.sh assumes it run on Travis

edgar-bonet, 2020-04-23:
Although the script attempts to support both Travis and GitHub actions, in some parts it still assumes it runs on Travis. For instance, [line 81][]:

```shell
git config user.email "travis@travis-ci.org"
```

and [line 168][]:

```shell
git commit -m "Deploy code docs to GitHub Pages Travis build: ${TRAVIS_BUILD_NUMBER}" -m "Commit: ${TRAVIS_COMMIT}"
```

This results in commits like [this one][commit]:

```text
commit 57834c85e08e3dd436743a59b72f2d888dafd2c5
Author: Doxygen CI <travis@travis-ci.org>
Date:   Thu Apr 23 17:51:50 2020 +0000

    Deploy code docs to GitHub Pages Travis build:
    
    Commit:
```

[line 81]: ../blob/37dc6a11467f25e26f81db2022e60473b2e69d1f/doxy_gen_and_deploy.sh#L81
[line 168]: ../blob/37dc6a11467f25e26f81db2022e60473b2e69d1f/doxy_gen_and_deploy.sh#L168
[commit]: /adafruit/RTClib/commit/57834c85e08e3dd436743a59b72f2d888dafd2c5

ladyada, 2020-12-29:
feel free to submit a PR

edgar-bonet, 2020-12-30:
I just submitted pull request #91 addressing this issue.

commit 4fbe599c (2021-01-03): Format documentation commit in GitHub actions

The commit message provides the SHA1 of the current commit and an ID for
the build, now both on Travis and GitHub actions.

Fixes #74.
closed 2021-01-03 by ladyada