adafruit/ci-arduino #84: Testing examples of my library for ESP8266 fails.

beegee-tokyo, 2020-09-06:
Since some time testing for ESP8266 fails for me with the following debug output:
```log
----------------------------- DEBUG OUTPUT -----------------------------

Picked up JAVA_TOOL_OPTIONS: 

Set log4j store directory /home/travis/.arduino15

Loading configuration...

Initializing packages...

Preparing boards...

Verifying...

fork/exec /bin/xtensa-lx106-elf-g++: no such file or directory

------------------------------------------------------------------------
```

I didn't change anything in my [.travis.yml](https://github.com/beegee-tokyo/SX126x-Arduino/blob/master/.travis.yml).
But the error is always the same. Testing against ESP32 and nRF52 works fine.

Anybody has an idea?

per1234, 2020-09-06:
Try clearing the cache on Travis CI:
https://docs.travis-ci.com/user/caching/#clearing-caches

beegee-tokyo, 2020-09-07:
@per1234 
Clearing the cache helped. Thank you.

closed 2020-09-07 by beegee-tokyo