adafruit/ci-arduino #90: Github build will break soon, fails on Ubuntu 20.04

zjwhitehead, 2020-12-28:
Currently the actions_install.sh script fails due to unmet llvm requirements in Ubuntu 20.04 
[image]

The `example_actions.yml` as well as many other libraries [specifies](https://github.com/adafruit/ci-arduino/blob/master/example_actions.yml#L7) `runs-on: ubuntu-latest`

[image]

[Github indicates](https://github.com/actions/virtual-environments/issues/1816) the transition to 20.04 for ubuntu-latest is already happening which means more builds will start breaking soon. 

## Potential fix
changing in `actions_install.sh` to use `libllvm8=1:8.0.1-9 -V` seems to work but needs further testing. Also its not backwards compatible with 18.04 so further work will be required to update the script to handle both versions.
[image]

**Bonus:** For now pinning the example config to `ubuntu-18.04` is probably worth it...

ladyada, 2020-12-29:
hmm! can you submit a PR? we'll try it out :)

dhalbert, 2021-03-03:
Stopgap fix: https://github.com/adafruit/Adafruit_Learning_System_Guides/pull/1474 for now. We should update the whole `clang-format` checker.

ladyada, 2021-05-05:
this was fixed i guess cause it works now!

closed 2021-05-05 by ladyada