adafruit/ci-arduino #148: Not working with Arduino IDE version 2.x

Aypac, 2023-01-11:
It looks like the format for downloading is `https://downloads.arduino.cc/arduino-$ARDUINO_IDE_VERSION-linux64.tar.xz`, where it appears that since version 2.0.0 there is only `https://downloads.arduino.cc/arduino-ide/$ARDUINO_IDE_VERSION_Linux_64bit.zip`. I guess that needs an additional check.

per1234, 2023-01-12:
Hi @Aypac. [Unlike Arduino IDE 1.x](https://github.com/arduino/Arduino/blob/master/build/shared/manpage.adoc), Arduino IDE 2.x does not have a command line interface, so it could not be used with the [`install.sh`](https://github.com/adafruit/ci-arduino/blob/master/install.sh) script maintained in this repository even if the script was updated to be able to download Arduino IDE 2.x.

However, this is not a problem because Arduino created a dedicated command line tool named [**Arduino CLI**](https://arduino.github.io/arduino-cli/latest/) that has all the functionality of Arduino IDE 1.x's CLI and much more. **Arduino CLI** actually provides all the non-GUI functionality for Arduino IDE 2.x, so using Arduino CLI in your CI system provides effective coverage of Arduino IDE 2.x compatibility.

The modern [`build_platform.py`](https://github.com/adafruit/ci-arduino/blob/master/build_platform.py) Python script hosted in this repository is already using Arduino CLI, so you only need to switch your CI system from using `install.sh` to `build_platform.py` (there is also an associated [`actions_install.sh`](https://github.com/adafruit/ci-arduino/blob/master/actions_install.sh) script to install the tools if you like) and you will be using the modern Arduino tooling. See the [readme](https://github.com/adafruit/ci-arduino#readme) for instructions.

Aypac, 2023-01-14:
Thank you @per1234 !
In the meantime I actually discovered that Travis CI is not free for open-source projects anymore. So I decided that rather than repairing, I'd move over to Github "Actions". They actually seem a bit easier to setup anyway and I find it nice to have it all in one place. Plus they are free for Open-Source (for now). I see you also added an according note on the tutorial on the adafruit website.

closed 2023-01-14 by Aypac