adafruit/ci-arduino #165: Ensuring the CI process is documented for Adafruit Learning System guide authors

TheKitty, 2023-10-04:
Per @ladyada request tag:

The Arduino-CI repo changed significantly from when the user guide was done with some new yml files, etc. The original guide was requested by you to help Learn authors know about .platform.test.only files. Earlier this year things had changed enough that the guide was out of date and when Melissa was having Qualia PR issues I compared things and noticed the discrepancies which I updated.

Here is the current guide page, updated 10/3/23 to align with the right yml file for author reference (it is hoped). https://learn.adafruit.com/admin/guides/3445/editor/23085

I am hoping to get some a repo maintainer to ensure I had captured the pertinent data and to double check the instructions in the README.md were also up to date. This may help folks going forward in getting the right .test.only file in future guides.

TheKitty, 2023-10-04:
@ladyada tagging again per Slack request

ladyada, 2023-10-04:
reviewed! made one change, to remove the line number on the CI script as it will almost certainly change

closed 2023-10-04 by ladyada