adafruit/ci-arduino #169: CI is grabbing ESP32 pre-releases

ayushsharma82, 2023-10-25:
So the issue is that ci-arduino which is used for continuous integration for my ESP32 libraries is failing because it is grabbing the latest alpha2 pre-release of ESP32 board. This in return causes the library CI to fail as there are bugs/changes in that release which won't let me compile my library.

How can I force it to use a specific board version?

ladyada, 2023-10-25:
we do this on purpose so we can fix libraries ahead of the release - pre-releases become releases pretty fast! you can fork/edit the installer if you need a different BSP version

ayushsharma82, 2023-10-25:
> we do this on purpose so we can fix libraries ahead of the release - pre-releases become releases pretty fast! you can fork/edit the installer if you need a different BSP version

Gotcha. I tried looking through the source code but it's very convoluted for a person who hasn't worked on it before. Can you refer to line/lines which can be modified to set specific BSP version?

ladyada, 2023-10-25:
https://github.com/adafruit/ci-arduino/pull/168

closed 2023-10-25 by ladyada